cclrobotics/lhr-testing #1: Check limit switches

griswaldbrooks, 2016-07-17:
_From @griswaldbrooks on July 17, 2016 0:14_

Verify that all of the limit switches work.
Some of the soldering is suspect.

_Copied from original issue: cclrobotics/lhr-docs#10_

ddigor, 2016-07-18:
Given the wear of transport and the previous marginal cable management,
there may be problems with some of the connectors as well as the solder.

How shall we denote which limit switch is which?
There is one for the two synchronized stepper motion axis  call it X
(really there should be a limit switch for each belt track),
one for the other motion axis, Y,
one for pipette position, Z,
perhaps one for the pipette piston.

griswaldbrooks, 2016-07-18:
I've already seen broken solder joints so each switch and cable assembly needs to be repaired and tested with a multimeter before integrating into the system.
To be consistent with the legacy documentation, I think we should continue to denote the axis with the two steppers the Y axis, the other motion axis the X axis. The Z is as you said. When talking to Pedro on 7/16/16, he referred to the pipette piston axis as the A axis.

I might agree with you about the needing an additional limit switch for the other belt track, though Pedro assured us when we asked him that it was never a problem, and as long as the belt teeth don't slip, I'd agree with him. But of course if it does slip...
So that should be a potential safety upgrade.
I'm also a proponent of adding a max limit switch in addition to the min limit switch, which the smoothieboard will accept.

ddigor, 2016-07-29:
I tested and repaired the 4 limit switches.
The Z axis switch needed no repair.
Solder, electrical continuity, heatshrink for insulation and strain relief were all intact.
I re-soldered and added new heatshrink on the X, Y and A axis switches.
The color coding is the same on all 4 switches.
Yellow - Common
Black - Normally Open
Red -  Normally Closed

griswaldbrooks, 2016-07-29:
Great! Should we close this ticket then, or is there more testing to do?

ddigor, 2016-07-29:
I think we can close it.

closed 2016-08-21 by griswaldbrooks