cclrobotics/lhr-testing #12: Electrophoresis box doesn't work.

griswaldbrooks, 2016-07-17:
From #9, plugged in the electrophoresis box with power supply and used multimeter to check output voltage.
Saw something in the order of 20 Volts (could have been less or more, not sure) and turned the adjustment knob. The voltage changed and then a loud screech came from the box and didn't stop until power was disconnected.
When turning the box back on, there was no screech and the voltage output was 23.8 V.
@tamarum says the voltage should be near 200 V.
@ZenSpiders is going to contact Pedro or Luis about the issue.

ddigor, 2016-07-29:
I looked at the components inside the clear green plastic power supply box.
It is a simple DC to DC step up converter built around the Maxim MAX1771,
which is designed specifically for such converters.
The components are well marked and correlate well with the reference design in the MAX1771 datasheet. From the failure symptoms, the most likely failed component is the power N-FET,
probably a Fairchild IRFS440, the next most likely the MAX1771 controller.
Given that something close to the 24V DC input appears at the output now, it's probably not the 100uh inductor or the Schottky diode.
The parts are cheap.
Repair will be a bit unpleasant, given the high voltage, and given that switching power supplies do almost nothing when they are not working.
[MAX1771.pdf](https://github.com/cclrobotics/lhr-testing/files/391280/MAX1771.pdf)
[IRFS440B.pdf](https://github.com/cclrobotics/lhr-testing/files/391288/IRFS440B.pdf)

griswaldbrooks, 2016-07-29:
Having not looked at the circuit, I assume it would be hooked up similarly to Figure 8 in the MAX1771 datasheet? I hope they're not applying the 24V directly to the MAX chip, as it's not rated for anything higher than 16V.

ddigor, 2016-07-29:
My examination was cursory, through the green plastic case in situ, with flashlight assistance,
but I didn't see any regulator for a lower supply to the MAX1771.
Maybe we were supposed to run the DC to DC converter off of 12V.
I will do a full extraction of the circuit, unless someone else gets there first.

griswaldbrooks, 2016-07-29:
That could be a reason the circuit blew. We won't be there tomorrow so if someone does it won't be me.

ddigor, 2016-08-03:
I examined the electrophoresis power supply more carefully.
It expects a 15V 1.6A power source.
24V is indeed too much.
The most important discovery was to remove the box from the foamcore
and look at the back, where the manufacturer and PCB revision were printed on the board.
   http://public.iorodeo.com/docs/electrophoresis_power_supply/
The board does seem to match the schematic closely,
although it says rev 1.4, while the documentation is rev 1.3
I need to take one more look at the numbers on the plastic case
(do they really go up to 200V?)
and the values of R2 and RV1, which would affect maximum voltage and adjustment range.
RV1 seems unmarked, so I'll need to measure it.
In any case a 40% over-voltage seems likely to have destroyed the MAX1771.
C2 and C4 were also over their rated voltage, but are more likely to have survived.
I will order parts and a more appropriate power supply.

The IO Rodeo board acknowledges being based on
   http://desmith.net/NMdS/Electronics/NixiePSU.html
which has more design analysis.

griswaldbrooks, 2016-08-03:
Guess I should have removed it from the foamcore before. Great to know where it comes from. I don't know anything about what the chemists need personally but @tamarum says 200V is common. The iorodeo site says it should go to 100V.

ddigor, 2016-08-04:
I re-examined the IO Rodeo box.
The numbers on the dial are 50 to 200V.
The resistors on the board appear to match the schematic,
and by design should produce 25 to 100V.
Perhaps R3 15K ohm is actually populated as 7.5K ohm - hard to tell through the green plastic.
The design is probably capable of 200V; 
the Nixie tube supply the design it was closely based on output 150 to 220V.
Parts ordered for most options.
  debug order plan
     1 - just try the 15V supply (unlikely)
     2 - replace MAX1771, 20V caps
     3 - replace FET
     4 - change R3 to 7.5K (if necessary for 200V)

griswaldbrooks, 2016-08-04:
Can the green box be easily disassembled? I think the 7.5k plan should probably work. I've not yet seen in the datasheet why it shouldn't be able to handle this voltage as the transistor is rated for it, but I agree, it's better to get the system working first with the stock configuration and then we can upgrade it using the 7k5.

ddigor, 2016-08-04:
The green box is actually pretty easy to take apart.
I had it apart for my careful confirmation, but I was in a rush for me second re=examination.

ddigor, 2016-08-06:
I've taken the green box to repair at work.
Back on Tuesday 8/9

griswaldbrooks, 2016-08-06:
Great! We were wondering where it was.

ddigor, 2016-08-10:
I replaced the MAX1771, powered the box with a 15V 1.6A supply,
and it worked, output adjustable between 28V and 102V,
pretty much according to the documentation.
I changed the 15K resistor to 7.5K, 
and the output now is adjustable between 31V and 204V.
I think we can close this.

griswaldbrooks, 2016-08-10:
@ddigor, I started a quick readme for the [Electrophoresis System](https://github.com/cclrobotics/lhr-docs/tree/master/System-Diagrams/Electophoresis-System). Do you think there's anything else from this ticket that should make it's way over there? Obviously this doc needs revision.