cclrobotics/lhr-testing #4: Read the temperature sensors

griswaldbrooks, 2016-07-17:
_From @griswaldbrooks on July 17, 2016 0:7_

Load a basic sketch onto the Arduino to read the temperature/humidity sensor.

_Copied from original issue: cclrobotics/lhr-docs#8_

ddigor, 2016-08-07:
DHT library appears to be from Adafruit
   https://github.com/adafruit/DHT-sensor-library

griswaldbrooks, 2016-08-07:
Correct, I'm more concerned with making sure the continuous integration stuff can compile everything and update/get the correct version when someone else wants to use this.

ddigor, 2016-08-08:
I was confused by the DHT library location, but it makes sense now.
lhr-equipment-controller test code has DHT commented out, so no need for the library yet.
lhr-testing has the DHT library, but no code in igor-firmware yet.

griswaldbrooks, 2016-08-08:
Currently, the `lhr-testing/controller-tester/controller-tester.ino` has the `DHT` library commented out. Right now it has no need for it, but it obviously will because we need the test code to be able to read the sensors.
The `lhr-equipment-controller` repo looks weird because the `DHT` library AND `igor-firmware` are actually different github repos and our repo knows this so it won't let us go into them. They're not empty.
The igor firmware is here: https://github.com/arcturusbiocloud/igor-brain
If you clone the repo and look at it's remote, you see it's the above. I don't like the arrangement because it means that for us to make changes to our controller firmware we'd have to push to Pedro's repo. Our firmware is going to change significantly from this anyway so I'm going to split it off in the near future (ie, remove the `.git` folder from the `igor-firmware` folder and start changing things).

As for the `DHT` library I'm not sure the most maintainable thing to do is have this cloned repo in our repo. It might be the best option, because Arduino, but I'd prefer to make it an installation dependency, something akin to this: https://learn.adafruit.com/continuous-integration-arduino-and-you/testing-your-project

But again, I'm not sure what the best strategy is here. Also, I hear `git submodules` are actually a small nightmare so I won't go in that direction.