cclrobotics/lhr-testing #5: Test the Smoothieboard

griswaldbrooks, 2016-07-17:
_From @griswaldbrooks on July 17, 2016 0:6_

Do a basic hookup of the Smoothieboard with some stepper motors and send some commands.
Most likely use Pronterface to run the motors.

_Copied from original issue: cclrobotics/lhr-docs#7_

ddigor, 2016-07-29:
There are three different Casun stepper motors, with different current settings
Online data is funky, but this is what I found so far
17HD202-C3
  X and Y axes
  http://www.smoothmotor.com/nema-17hd-stepper-motor
  0.5A 12.5V  25ohm
42SHD0217-250NJ
  Z axis
  http://www.casunmotor.com/nema-17-stepper-motor
  1.5A 3.75V 5.0ohm (seems off by a factor of 2)
42SHD0001-02TT
  A axis
  http://www.casunmotor.com/nema-17-stepper-motor
  0.4A 12V 30ohm
Given that the Smoothieboard is a current limiting controller,
powered by 12V in the present design,
only the Z-axis may be capable of high step rates with good torque.
Changing the Smoothieboard to 24V supply might improve torque at speed.
All of which might not matter.
We should be sure the current settings 
in the Configuration File on the Smoothieboard SD card 
do match the motors.

griswaldbrooks, 2016-07-29:
Agreed. We'll have to pull the smoothieboard config. Seems strange that the weaker motors are being used for moving the bigger load, though I guess there are two motors for the Y-Axis.

tamarum, 2016-08-07:
I tested the SmoothieBoard using one of the Z-axis stepper motors (from the side meant for the multi-channel pipetter). 

It worked fine, however this side does not have a designated limit switch so at some point the motor pulled one of the acrylic platforms towards itself. It not stopped it could wear away the threads on the acrylic piece. 

I used the latest [compiled version](http://koti.kapsi.fi/~kliment/printrun/) of PrintRun. It contains Pronterface, which was pretty intuitive but didn't operate the motor very smoothly. In the future though I would prefer we write our own controls for it to allow us to control the CNC using the keyboard and smoother motions. (Arrow keys for X-Y, and other keys for the two z and a axes). It would be even better if we could get some automated protocols set up.

I'll make more tickets for documentation to include the information Donald found and shed some light on how we should use Pronterface, what connects where, how to read/write the smoothieboard config file, etc. I'll also make one for testing all of the motors, perhaps next week.

tamarum, 2016-08-07:
Note: I connected the Smoothieboard to my Mac to do this test. I did **not** go through the raspberry pi.

closed 2016-08-07 by tamarum