cclrobotics/lhr-testing #7: Test the relays

griswaldbrooks, 2016-07-17:
_From @griswaldbrooks on July 17, 2016 0:3_

Load a basic sketch onto the Arduino and switch the relays.

_Copied from original issue: cclrobotics/lhr-docs#5_

tamarum, 2016-07-24:
## Tested shaker relay
Made sure the shaker power adapter was wired to a relay circuit, plugged into an outlet, and connected to the shaker. I connected the power and ground pins from the relay circuit to VIN and ground pins on the Arduino, respectively. I connected the signal receiving pin on the relay with a VIN pin on the Arduino and powered the Arduino. This essentially sent a "HIGH" signal to the relay which turned on the shaker. I then connected this signal wire to the Arduino pin designated for the shaker (according to the Arduino code @griswaldbrooks pulled off of the raspberry pi which can be found [here](https://github.com/cclrobotics/lhr-equipment-controller/blob/master/igor-firmware.ino)). We were able to then open a serial terminal through the Arduino IDE and power on/off the shaker. The other relays should work in a similar manner, and all of the relays will be connecting to the Arduino through a shield. 

Started a new issue to create a general diagram of the relays and Arduino shield work here: cclrobotics/lhr-docs/15

griswaldbrooks, 2016-08-07:
Copied the controller firmware to the testing repo and modified it to take manual commands for the incubator pins.
Tested all the relays without AC connected to them and they all switched on and off as expected. The ports tested are:
D2, D3, D4, D5, D6, D7, A1, A2
Which is all of the known relays. They were tested using the arduino and the interface shield.

closed 2016-08-07 by griswaldbrooks