cclrobotics/lhr-testing #9: Power on check the system components.

griswaldbrooks, 2016-07-17:
_From @griswaldbrooks on July 17, 2016 0:0_

Need to verify that the system components work.

Power on checks:
- [ ] Raspberry Pi
- [ ] Arduino
- [ ] Smoothieboard
- [ ] Electrophoresis voltage regulator
- [ ] OpenPCR
- [ ] Centrifuge
- [ ] Shaker
- [ ] Freezer
- [ ] Freezer arduino
- [ ] UV Lightbox
- [ ] VL Lightbox
- [ ] Incubator

_Copied from original issue: cclrobotics/lhr-docs#3_

griswaldbrooks, 2016-07-17:
Raspberry Pi powered on with issue. Went to boot screen,

griswaldbrooks, 2016-07-17:
Arduino powers up fine and drivers install on Windows computer.
Shaker and centrifuge turn on fine. None of these were done with their relays.

griswaldbrooks, 2016-07-17:
Electrophoresis box "powers up" but doesn't work. #12 describes the issue.

griswaldbrooks, 2016-07-17:
Smootheirboard powers up. Windows recognizes it, though says it needs a different driver.

griswaldbrooks, 2016-07-17:
The arduino for the freezer is an Arduino Pro Mini.
After plugging it into its power supply, the red power led comes on and the green LED 13 comes on.
It needs an FTDI to communicate with it which we don't have on hand.

tamarum, 2016-07-17:
Freezer works. Temperature at the bottom of the fridge was -20˚C, near the opening it is -12˚C. Pedro suggested putting a jack in the freezer to raise samples to near the top of the fridge to allow them to thaw. We should look into this in the future.

griswaldbrooks, 2016-07-17:
Both OpenPCR machines turn on. Both lightboxes power on.

griswaldbrooks, 2016-07-17:
The incubator cannot be tested. It is most likely just a lightbulb, which we cannot find.

closed 2016-07-17 by griswaldbrooks