cclrobotics/lhr-testing #15: Testing Smoothieboard w/ multiple motors & limit switches

tamarum, 2016-08-07:
Test axis and plunger motors and their limit switches.
(Won't test the multi-channel side)

tamarum, 2016-08-21:
Tested the motors. They work, however there was an issue with the X-axis motor connection. 
Griz fixed the connector and now it seems to work properly. 
The X, Y and Z operate fine through Pronterface.

I'm sure sure how to control the A and B axes but I'm guessing this is done by sending g code commands through the Pronterface serial communication and by loading a full g code script.

Also tested the limit switches according to the smoothieboard website through Pronterface. They did not seem to do anything, the motor just continued to run into the switch without stopping. We also tested them by pressing the switch while the motor was running, in case they were not physically being reached. They did not stop the motor. We might not be using them correctly, will further look into this.

Through this process I saw that some of the connectors are loose, have some weak soldering, and needed fiddling with to ensure a good connection between the smoothieboard and steppers. I'll make another issue for replacing the connector wiring.

ddigor, 2016-08-28:
One of the two wires between the Y connector harness and the motors, 
on one of the connectors that mates with the cable Y,
the red wire appears broken.
The cable Y connector need to be documented.
There are a few different wiring patterns at the single end of the Y
that would perform the phase inversion for the two oppositely facing motors.
We need to document the pattern actually chosen.
At all connector junctions other than the single end of the Y at the Smoothie board,
Y to cables, cables to motors.
the wire colors - red, yellow, green, black -
are the same on both side of the connectors.
What is our naming convention for the two Y-axis motors?

tamarum, 2016-08-28:
I made a [diagram](https://github.com/cclrobotics/lhr-docs/blob/master/System-Diagrams/Motion-Axes/README.md) that gives an overview of how the motors/limit switches are set up.
Griz started replacing all the weak/improperly connected wires with new crimps in the connectors.
He also connecting the two Y motors using a small perf board instead of forcing two sets of wires into one connector. Unfortunately we realized there's one connector that will go in this perf board that we didn't get the right crimps for, which is the same one the broken red wire goes to. We'll order some for next week. I will make an amendment to the aforementioned file to describe the perf board.

griswaldbrooks, 2016-08-28:
The designators should all end in a number (eg M-X1) to be consistent.
The diagram view labels should all be in the same place in each diagram.